AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule #4: Missing BOOTRST fuse in preparation instructions

sgmne, 2016-08-04:
For the boot loader to work properly the BOOTRST fuse must be turned on.

Currently the fuse values and screenshots in the preparation instructions
(01. Preparing atmega* Xplained mini for Arduino IDE use) show that fuse
to be off.

aethaniel, 2016-09-05:
Hi @sgmne, thanks for your comment. The AVR datasheets are not very clear on the use of this fuse and I'm not expert on these devices. This may impact also https://github.com/AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule/blob/master/module/boards.txt as I updated all fuses information end of July for release https://github.com/AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule/releases/tag/v0.4.0

sgmne, 2016-09-05:
Hi @aethaniel. The description of the BOOTRST bit is in the section 32.5 of the ATmega328PB datasheet. Its behavior has not changed from ATmega328P which is used in Arduino Uno and thus has to have the same state for the optiboot to work correctly in Xplained Mini. Without BOOTRST on the board can be programmed only once with Arduino IDE (avrdude) and then the memory has to be cleared with Atmel Studio and the boot loader reprogrammed for next programming with Arduino IDE. With BOORST on the board behaves like any Arduino and the user's application can be reprogrammed with Arduino IDE without reprogramming the boot loader with  Atmel Studio.

I think the flags were correct in a previous version in May when I first tried this out.

aethaniel, 2016-09-06:
Thanks @sgmne, the issue is fixed in https://github.com/AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule/commit/ffab7b7c25d60809befa6fcd678031bb278b49a8 and https://github.com/AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule/commit/aaf5782980ca45954c7ebd9ecf8827ea75ff6015.

A new release (https://github.com/AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule/releases/tag/v0.5.0) is out, will try to push it to Arduino server.

closed 2016-09-06 by aethaniel