AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule #8: Programmer not responding

aldenderfer, 2016-12-03:
After following your walkthrough - setting the fuse bits in Atmel Studio and programming the bootloader - I'm getting a timeout when attempting to actually program my 328pb Xplained. It's a standard "not in sync" error (resp=0x00). The board shows up in ports and reads properly (well, as a 328p, but I assume that's fine). Could this be a problem with the board defs in boards.txt?

Setup:
-- ATmega328pb Xplained, firmware v1.13
-- Atmel Studio 7.0
-- Arduino 1.6.13 (programmer: Atmel mEDBG

aethaniel, 2017-03-23:
Hi @aldenderfer, sorry, I don't receive notifications when I'm not explicitly named in a comment :-(
Is your problem still present?

aldenderfer, 2017-03-28:
@aethaniel No worries! I will remember this for the future.

I resolved this shortly after posting, but I can't recall how - sorry. This can be closed!

closed 2017-03-28 by aldenderfer